DFRobot/DFRobot_BME680 #1: No Wire.endTransmission() needed after Wire.requestFrom().

Koepel, 2017-09-06:
In the file "DFRobot_BME680.cpp" the Wire.requestFrom() is followed by a Wire.endTransmission().

That Wire.endTransmission() can be removed, since it should only be used for writing data.

forKnowYou, 2017-09-20:
Sorry for late reply.
It's my pleasure to receive your advise.
We have tested and found that the library cannot read registers when the sentence been removed.

Koepel, 2017-09-20:
I mean the last `Wire.endTransmission();` in the function readReg(), not the one that is used to write the register address to the chip.

Are you saying that reading the registers **before** that line can not be done if that line is removed ?

The Wire.endTransmission() should only be used when writing data. It should not be used after a Wire.requestFrom(). That is how the Wire library works. That is how the documentation is.

In case you did remove the right line: It will cause an extra delay, perhaps that is needed. When the Wire library for AVR chips is used, a Wire.endTransmission() on its own puts the sensor address on the I2C bus and the sensor acknowledges that.

forKnowYou, 2017-09-20:
You mean the line 106 in DFRobot_BME680.cpp right?
We had try to removed it, and it looks like BME680 need that single.
I am a yong engineer, thanks for your explanation. I will do more exploration.

Koepel, 2017-09-20:
No, line 112.
That is **after** the Wire.requestFrom().

forKnowYou, 2017-09-21:
Understood, thanks.

Koepel, 2017-09-22:
I see the issue has been solved. I will close this issue.

closed 2017-09-22 by Koepel