DFRobot/DFRobot_BME680 #4: Altitude Correct?

pedda4711, 2017-10-16:
now i've compiled the example sketch. an load it to my esp8266 12e...

Is there something wrong in calculate the altitude?
I get a negative value. (My real altitude is near by 120m) 
(The gas value went higher after a few minutes. Up to 400 ppm in my living room. )

temperature(C) :21.44
pressure(Pa) :101698.00
humidity(%rh) :64.53
gas(ppm) :242
gas resistance(ohm) :126410.00
altitude(m) :-31.01

Greetings, Peter

ouki-wang, 2017-10-17:
@pedda4711 thanks for your test
I also found this problem, I will fix it。

forKnowYou, 2017-10-19:
Thanks for your feedback , I'm so sorry for that problem.
Please download the new lib and feedback.Thank you.

pedda4711, 2017-10-20:
Now it works :)

closed 2017-10-20 by pedda4711