DFRobot/DFRobot_BME680 #5: Gas sensor values

kylix34, 2017-10-18:
Hello,

I'm getting pretty high values from the gas sensor: somewhere between 90-980 ppm in the living room.
Is it normal?

Bosch datasheet talks about the IAQ Index Air Quality between 50-500:

0 – 50 good
51 – 100 average
101 – 150 little bad
151 – 200 bad
201 – 300 worse2
301 – 500 very bad

At 980ppm I suppose I should be dead :D

MariusRumpf, 2017-10-18:
I am getting the same results indoor here with some, for me unreasonable jumps over time.

[image]

janzieg, 2017-10-18:
Same for me here (using https://github.com/watterott/BME680-Breakout) - i know about other voc sensors (iaq core, ccs811) that it might take some time (about 2 days) until automatic calibration (baseline correction) is on a good level which results in "realistic" values. Maybe the BME680 needs some more time until calibrated...?

MariusRumpf, 2017-10-18:
@janzieg The graph from above is from a sensor running for about 3 weeks in the same place. It has only been restarted for a software update with a new version of the library a week ago. I have been using the same breakout as you mentioned. I have seen this drops since the beginning with old and new versions of the library.

janzieg, 2017-10-18:
@MariusRumpf  Have you tried other drivers libs yet ? The sensor is not really usable like this.
As far as I could see, the original bosch library was integrated in this lib, so it supposed to be correct (even not guaranteed). Maybe the board is not designed correctly ?

MariusRumpf, 2017-10-18:
@janzieg I have not tried other libraries and only used I2C. Maybe it's the board... I don't know. 

Feel free to ask if more details are needed.

kylix34, 2017-10-18:
This is my graph for one week (I have the sensor running for more than 2 weeks now):
[image]

forKnowYou, 2017-10-19:
@kylix34 , @MariusRumpf 
Thanks for yours feedback.
That gas result is not normal, so we delete it and please use gas resistance to statistics air quality.
I am sorry for the inconvenience caused.

BlueDot-Arduino, 2017-10-20:
Please note that the gas resistance is not the same as the IAQ index. 
The BSEC algorithm from Bosch uses the measured gas resistance to calculate the IAQ index.
The problem is that the BSEC algorithm is not so easy to read. 
While the sensor API can be downloaded from Github or from the website from Bosch Sensortec (all those .c and .h files), the BSEC files consist of .a and .lib files.
So far I didn't manage to open them, but the formulas to calculate the IAQ index are written there.

janzieg, 2017-10-20:
@BlueDot-Arduino Thanks for your efforts! So the BSEC library (with closed source) cannot be included "as it is" - as precompiled binary ?

BlueDot-Arduino, 2017-10-20:
I'm not sure, I haven't tried that yet. But apparently, the BSEC library "as it is", is only compatible with 32 bit MCU platforms (according to [Bosch](https://www.bosch-sensortec.com/bst/products/all_products/bsec)).

That would mean, we could use the BSEC library from Bosch, but only with 32-bit microcontrollers, like the Arduino Due, and not with the Uno. But I'm speculating.

wklenk, 2017-10-20:
I am currently trying to get this BSEC library working with a STM32F103C board ("Blue Pill") which has an ARM Cortex M3. I already managed that the linker in the Arduino IDE is no longer complaining about missing functions and symbols, but when I flash the software to the board just nothing happens. Any hint is appreciated :)

sebswed, 2017-11-02:
I have an Arduino Due and the BME680 on I2C. I'm not that skilled but I'm more than willing to test any code if that is of any help.

AlesSt, 2017-11-07:
Hi all!
Made an acc here just to answer here on this topic :)
Have been testing the watterott BME680 sensor for 2-3 weeks now and today I got VOC working (thankgod i havent upgraded library to the last one :)
VOC works with previous version of library IF you follow this - make measurements every second.
I incorporated it at the end of each loop with delay 1000 (MUST be at least 1000 = 1s) and after a minute or so it works (meaning measurements are not jumping all over the place) as it should.
I guess this is because the plate in sensor must be warm the whole time for the algorithm to calculate resistance correctly.........
@forKnowYou.......perhaps you should heat the plate longer? or once longer and next one as it is now and then return the result to the command
if there are some commands I could use in my sketch before the measurement (to heat the plate for 100ms, for 200ms...., to find out the time it needs to be heated) i can test and find the right procedure so u can implement bme.readGas() again?

ouki-wang, 2017-11-15:
I have got correct IAQ using FireBeetle-ESP8266.  
You have to wait 300 seconds for the bmp680 to heat up.
[image]

sebswed, 2017-11-15:
There is an interesting topic on this on https://forum.fhem.de/index.php/topic,78619.60.html that I have been following with google translate that may be of help.

closed 2017-12-27 by kylix34