DFRobot/DFRobot_BME680 #6: Minor Issue - begin() should return true when successful

janzieg, 2017-10-18:
It just took me some time to realize, that the v2.0 version returns _false_ after _successful_ initialization when _true_ is expected.
This was quite confusing for me...and took me some time (as I thought having a Connection issue).

closed 2017-10-19 by ouki-wang

janzieg, 2017-10-19:
@forKnowYou but this is also wrong in code (begin returns _false_ on success right now).

ouki-wang, 2017-10-20:
@janzieg I have fixed the bug。thanks for your support

Flowm, 2017-12-01:
It seems that after this was corrected in e06b5772275ffc613e0ced21dd532893faadac4c it was changed back in https://github.com/DFRobot/DFRobot_BME680/commit/f2a0e0a31adff5771f95896fd66ca87b6bb86dcd#diff-e659cb54fd070752f5fedba933c0b0c1R66

https://github.com/DFRobot/DFRobot_BME680/blob/f858b4874abecc80a42aa136978b1faa69bcc629/DFRobot_BME680.cpp#L66-L68

Thus begin() now returns `true` on failure and `false` on success again.