DFRobot/DFRobot_BME680 #14: ESP8266 12E

darox, 2018-02-09:
Does this work for ESP8266 12E as well? 
I'm getting the following error:
ibraries\DFRobot_BME680-master\bsec_integration.c.o:(.text.bsec_iot_init+0x8): undefined reference to `bsec_init'

libraries\DFRobot_BME680-master\bsec_integration.c.o:(.text.bsec_iot_init+0xc): undefined reference to `bsec_update_subscription'

libraries\DFRobot_BME680-master\bsec_integration.c.o: In function `bsec_iot_init':

C:\Users\Dario\Documents\Arduino\libraries\DFRobot_BME680-master/bsec_integration.c:182: undefined reference to `bsec_init'

libraries\DFRobot_BME680-master\bsec_integration.c.o: In function `bme680_bsec_update_subscription':

C:\Users\Dario\Documents\Arduino\libraries\DFRobot_BME680-master/bsec_integration.c:131: undefined reference to `bsec_update_subscription'

libraries\DFRobot_BME680-master\bsec_integration.c.o:(.text.bsec_iot_loop+0x2c): undefined reference to `bsec_sensor_control'

libraries\DFRobot_BME680-master\bsec_integration.c.o:(.text.bsec_iot_loop+0x40): undefined reference to `bsec_do_steps'

libraries\DFRobot_BME680-master\bsec_integration.c.o:(.text.bsec_iot_loop+0x83): undefined reference to `bsec_sensor_control'

libraries\DFRobot_BME680-master\bsec_integration.c.o: In function `bme680_bsec_read_data':

C:\Users\Dario\Documents\Arduino\libraries\DFRobot_BME680-master/bsec_integration.c:304: undefined reference to `bsec_do_steps'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

darox, 2018-02-09:
EDIT: c:/users/dario/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Dario\AppData\Local\Temp\arduino_build_54549/IAQ_I2C.ino.elf section `.text' will not fit in region `iram1_0_seg'

C:\Users\Dario\Documents\Arduino\hardware\esp8266com\esp8266/tools/sdk/lib\libmain.a(app_main.o): In function `uart_div_modify':

(.irom0.text+0x64): undefined reference to `__wrap_register_chipv6_phy'

C:\Users\Dario\Documents\Arduino\hardware\esp8266com\esp8266/tools/sdk/lib\libmain.a(app_main.o): In function `uart_div_modify':

(.irom0.text+0x7d): undefined reference to `__wrap_register_chipv6_phy'

C:\Users\Dario\Documents\Arduino\hardware\esp8266com\esp8266/tools/sdk/lib\libphy.a(phy_sleep.o): In function `pm_wakeup_init':

(.irom0.text+0xb37): undefined reference to `__wrap_register_chipv6_phy'

c:/users/dario/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-w_pow.o):(.literal+0x14): undefined reference to `__errno'

c:/users/dario/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-w_pow.o): In function `pow':

w_pow.c:(.text+0xab): undefined reference to `__errno'

w_pow.c:(.text+0xba): undefined reference to `__errno'

w_pow.c:(.text+0x13f): undefined reference to `__errno'

w_pow.c:(.text+0x14e): undefined reference to `__errno'

c:/users/dario/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-w_pow.o):w_pow.c:(.text+0x1c2): more undefined references to `__errno' follow

C:\Users\Dario\AppData\Local\Temp\arduino_build_54549/arduino.ar(core_esp8266_main.cpp.o): In function `optimistic_yield':

C:\Users\Dario\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp:57: undefined reference to `printf'

C:\Users\Dario\AppData\Local\Temp\arduino_build_54549/arduino.ar(core_esp8266_main.cpp.o): In function `do_global_ctors':

C:\Users\Dario\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp:57: undefined reference to `printf'

C:\Users\Dario\AppData\Local\Temp\arduino_build_54549/arduino.ar(core_esp8266_phy.c.o):(.iram.text+0x8): undefined reference to `__real_spi_flash_read'

C:\Users\Dario\AppData\Local\Temp\arduino_build_54549/arduino.ar(core_esp8266_phy.c.o): In function `__wrap_spi_flash_read':

C:\Users\Dario\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_phy.c:266: undefined reference to `__real_spi_flash_read'

C:\Users\Dario\AppData\Local\Temp\arduino_build_54549/arduino.ar(sntp-lwip2.c.o):(.text.settimeofday+0x8): undefined reference to `sntp_set_timezone'

C:\Users\Dario\AppData\Local\Temp\arduino_build_54549/arduino.ar(sntp-lwip2.c.o):(.text.settimeofday+0xc): undefined reference to `sntp_set_daylight'

C:\Users\Dario\AppData\Local\Temp\arduino_build_54549/arduino.ar(sntp-lwip2.c.o): In function `settimeofday':

C:\Users\Dario\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/sntp-lwip2.c:63: undefined reference to `sntp_set_timezone'

C:\Users\Dario\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266/sntp-lwip2.c:63: undefined reference to `sntp_set_daylight'

C:\Users\Dario\AppData\Local\Temp\arduino_build_54549/arduino.ar(umm_malloc.c.o): In function `umm_info':

C:\Users\Dario\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c:991: undefined reference to `printf'

C:\Users\Dario\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c:995: undefined reference to `printf'

C:\Users\Dario\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c:1028: undefined reference to `printf'

C:\Users\Dario\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c:1050: undefined reference to `printf'

C:\Users\Dario\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c:1076: undefined reference to `printf'

C:\Users\Dario\AppData\Local\Temp\arduino_build_54549/arduino.ar(umm_malloc.c.o):C:\Users\Dario\Documents\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c:1076: more undefined references to `printf' follow

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module)

jttaylor86, 2018-03-28:
Hello,

Can I really use this library with an ESP32? There is no eagle.app.v6.common.ld in the esp32 sdk.

Thanks.

khongco44, 2019-01-22:
Hello
How to fix this issue

marcelitum, 2019-07-20:
any fix ?

juergs, 2019-08-26:
"How to fix this issue"
Install BSEC - Static Library  in Arduino. 
https://www.bosch-sensortec.com/bst/products/all_products/bsec

closed 2023-09-12 by darox