DFRobot/DFRobot_BME680 #15: BME680 stop replying

sisamiwe, 2018-03-01:
Hi,

I'm using your example IAQ_I2C and it work fine.
Nevertheless after about an hour the BME680 stops working. There is no feedback anymore or serial output.
Could you help?

forKnowYou, 2018-03-05:
hi,
ESP8266 can work after restart with no bme680 restart?
Is there no feedback with a certain time?

sisamiwe, 2018-03-05:
I don't unterstand your questions?

I'm using the BME680 on an ESP8266 (Wemos D1 Mini) driven by the IAQ_I2C sketch out of the repo here.
The ESP8266 is monitored by serial communication. Approx 1 hour the BME is communicating with the ESP which I can see in the serial commuincation. After that, there is no communication by the BME680 to the ESP, but the ESP is still alive.

sisamiwe, 2018-03-14:
I'm trying to decribe the issue more precise.

I'm using a Wemos D1 mini having a ESP8266 onboard. The BME680 is connceted via I2C. I'm using Arduino IDE 1.8.5 for compiling and flashing. 
As a first step I'm using the example IAQ_I2C from the repo to get also the IAQ.
Everything works fine for around an hour. After that, the Wemos D1 mini doesn't "talk" via the serial interface anymore. After resetting the Wemos (pressing reset button) it, it works again about an hour. 

I also tested a sketch from the BOSCH repo also providing the IAQ. Here the Wemos does work "forever". No communication loss. Nevertheless the BOSCH sketch is not usefull for my final application. I would like to use the DFrobot one, but as described it does not work mature.

Please support to get it work.

photogeen, 2019-08-07:
Same issue here! I modified the sketch to work with MQTT. Everything works for an hour. After that it stops. This problem is almost one year old?