DFRobot/DFRobot_BME680 #18: Measurements only work in loop

jttaylor86, 2018-04-04:
Hi,

I am using the IAQ_I2C Arduino Sketch which works well.

I am trying to extend this sketch so that I can send the IAQ values to Thingspeak after the measurement, however, right now the sketch isn't working as expected. The IAQ value is consistently 25, it's almost like no evaluation using the BSEC Algorithm happens.

Is the IAQ_I2C Arduino Sketch only designed to work on its own? The IAQ value is consistently 25, it's almost like no evaluation using the BSEC Algorithm happens. As soon as I comment out 

//thingSpeak(); 

then the IAQ evaluation happens again.

I would appreciate your help.

Thanks,

Justin

donatog, 2019-10-19:
Hi!
I'm having a similar problem.
In the sense that the BME680 sensor doesn't update the data when I sent them to ThingSpeak.
I tried a sketch with the readings are sent on the serial monitor and all works fine.
Please could you give me any help?
Thanks

bryn51, 2020-05-03:
I am not sure if its a complete answer, but IAQ takes longer than the other readings.
Thinking time I guess, because its just a calculation based on several of the other readings
You need to do this:
if(bme.isIAQReady()) {
        Serial.print("IAQ :");
        float iaq = bme.readIAQ();