DFRobot/DFRobot_BME680 #19: Unable to compile DFRobot_BME680_IC2 with Arduino IDE

ITGuyDave, 2018-10-23:
When trying to verify/compile via Arduino IDE 1.8.7 on Windows 10 Pro, the compiler fails every time on the BSEC integration component. The warning for the bme680.c also appear to be fixed in later versions of that library, suggesting the libraries used in this package are somewhat outdated and need refresh.

C:\My Documents\Arduino\libraries\DFRobot_BME680-master\bme680.c: In function 'calc_pressure':
C:\My Documents\Arduino\libraries\DFRobot_BME680-master\bme680.c:841:12: warning: left shift count >= width of type [-Wshift-count-overflow]

  var4 = (1 << 31);

            ^
C:\Users\Dave\AppData\Local\Temp\ccO4Xbmd.ltrans0.ltrans.o: In function `bsec_iot_init':
C:\My Documents\Arduino\libraries\DFRobot_BME680-master/bsec_integration.c:186: undefined reference to `bsec_init'

C:\Users\Dave\AppData\Local\Temp\ccO4Xbmd.ltrans0.ltrans.o: In function `bme680_bsec_update_subscription':

C:\My Documents\Arduino\libraries\DFRobot_BME680-master/bsec_integration.c:140: undefined reference to `bsec_update_subscription'
collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

The Adafruit BME680 libraries compile using the Bosch driver, but lack much of the functionality of the DFRobot libraries for this sensor (also a DFRobot model). I have tried swapping out the drivers, but I cannot determine where the bsec_integration.c library came from, as it's not in any of the builds on GitHub.

ITGuyDave, 2018-10-23:
I have tried updating the BME680 driver to 3.5.9 and also followed the steps for compiling BSEC from the project's GitHub Readme - and I now have a different error entirely, pointing right back to the DFRobot libraries:

C:\My Documents\Arduino\libraries\DFRobot_BME680-master\DFRobot_BME680.cpp: In member function 'int8_t DFRobot_BME680::iaqUpdate()':
C:\My Documents\Arduino\libraries\DFRobot_BME680-master\DFRobot_BME680.cpp:116:86: error: 'calc_gas_resistance' was not declared in this scope

     bme680_data.gas_resistance = calc_gas_resistance(gasADC, gasRange, &bme680_sensor);

                                                                                      ^
exit status 1
Error compiling for board Arduino/Genuino Uno.

rado99, 2018-12-04:
I face the same problem here. Did you fixed the problem? Any suggestions, because I am on the first step with a lot of errors.

bryn51, 2020-05-03:
Was this problem ever solved. (?)
I have the situation where the library compiles and links and runs OK (on a Arduino Mega 2560 with plenty of memory) using the example ino files, but when I add similar code to my own project, (after fixing all typos etc) I get errors similar to the above.  The error is caused by an unhappy linker, but its entirely unclear as to what is the cause.  I have tried lots of things, spending hours on the issue, but I'm flying blind.
At this point, I am chasing why the linker is unable to find bme_update_subscription.  That function appears to have gone missing.