DFRobot/DFRobot_BME680 #22: To get it working with Arduino (and ESP32S3)

zig-lh, 2024-01-05:
Hello,

i had compilation problem with BME680 and ESP32S3.
Here is the solution for me:
Library:
1.  install DFROBOT BME680 (2.0.0)
2.  install`` BSEC (1.8.1492)

Download this: https://community.bosch-sensortec.com/varuj77995/attachments/varuj77995/bst_community-mems-forum/12299/1/ESP32_S3_OUT.zip

Copy bsec_datatypes.h and bsec_interface.h in DFRobot_BME680 library's folder.

Create a folder esp32s3 in BSEC_Software_Library\src. 
Put libalgobsec.a in this folder.

In your sketch, add:
#include "bsec.h"

And it works.

Arya11111, 2024-03-07:
库中引用了bosch官方库https://github.com/boschsensortec/BSEC-Arduino-library，却代码不开源，在它支持的平台中不包含esp32s3
[image]

tobajer, 2024-10-11:
I used DFROBOT BME680 (2.0.0) and BSEC (1.8.1492) according to this hint, but it didn't compile. So I bring back the two original bsec_datatypes.h and bsec_interface.h, and it works fine. I run it on Arduino nano ESP32S3. Cheers.